adafruit/Adafruit_LC709203F #2: Confusing readings for cellPercent

caternuson, 2020-10-15:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=19&t=170429

Also recreated locally running library demo. Voltage agrees with multimeter reading. Ignoring temperature reading. Percentage seems odd though. Seems like it should be higher, assuming 4.2V to be the nominal 100%.
[image]

ladyada, 2020-10-15:
hmm ive plugged a battery in and am charging it, will see whats up, the chip is not very complex, probably some incorrect math or register setting. the percentage comes from a register read

DaveBKC, 2020-10-15:
Maybe worth noting: I didn't test this with CircuitPython, only Arduino.  Maybe CircuitPython math is ok?

DaveBKC, 2020-11-13:
Any news on this sucker?

ladyada, 2020-11-15:
yah turns out it defaults to thinking the battery is 4.35V instead fo the common 4.2V - latest commit fixes it to assume 4.2V and % should be more familiar!

closed 2020-11-15 by ladyada