adafruit/Adafruit_LC709203F #5: LC709203F returns 0 when attached to Adafruit MagTag?

ericklein, 2021-11-05:
- Arduino board: Adafruit MagTag. Adafruit Metro M0

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13

Using LC709203F_demo.ino (Adafruit example), lc.cellVoltage() and lc.cellPercentage() return 0 when attached to Adafruit Magtag. I've tested multiple LC709203F (Adafruit p/n 4712) boards with multiple Adafruit batteries (1000mA, 2000mA, 2500mA). The board is properly detected by the example code and I've adjusted lc.setPackSize() to the proper/closest mA battery size in each test run. I also have integrated the library and functions into my own code and the same 0 result is received.

To further examine the issue, I tested the same LC709203F boards and battery on a Metro M0 Express in the wiring configuration highlighted in the Adafruit Learning System example: https://learn.adafruit.com/assets/94624. This configuration works properly and returns values for lc.cellVoltage() and lc.cellPercentage().

The difference in wiring between the two configurations is:
- MagTag configuration: the battery is attached to the Adafruit LC709203F board via JST 2 wire and the MagTag is (properly) powered from the LC709203F board via the other JST 2 wire connector. The LC709203F board is also connected to the MagTag via StemmaQT connection.
- Metro M0 configuration: the battery is attached to the Adafruit LC709203F board via JST 2 wire and the LC709203F board is connected via Stemma QT breakout to 5V, GND, SCL, and SDA.

I reviewed the Adafruit Discord for any reported issues like this, and did not find any. I also reviewed the two issues posted in this Github repository and did not see this issue reported.

ladyada, 2021-11-05:
esp32s2 has had issues with i2c repeated start - its related - please check this PR as its not an arduino library issue but a board support package issue! :)

https://github.com/espressif/arduino-esp32/issues/4729

closed 2021-11-05 by ladyada

ericklein, 2021-11-05:
As usual, you are the best! thanks, getting data with BSP 2.0.1-RC1