adafruit/Adafruit_LC709203F #6: Would like better control in specifying battery pack size (APA)

disquisitioner, 2022-02-11:
I wish you hadn't made the battery pack size (APA) factor an enum (lc709203_adjustment_t) as that makes it impractical for me to use values other than what you've defined in the enum, even though I can easy interpolate more accurate APA values from the curve in the LC709203F datasheet.  (E.g., for a 1200 mAH battery pack an APA value of 0x1D would be more accurate than having to use the 1000 mAH value in the enum.).

disquisitioner, 2022-02-12:
Pull request submitted.

ladyada, 2022-02-12:
thanks! please check the CI output to get it ready for review and then we can merge :)

closed 2022-02-13 by ladyada

Seikonolff, 2024-07-31:
Hello there,

Bringing this back as I don't get why @disquisitioner 's setPackAPA function takes a 8 bits int as parameter while casting it to a 16 bits when using the writeWord function. 
For ex when I use it for a 5000 mAh battery the value is changed to 136 (5000%256).