adafruit/Adafruit_LC709203F #13: Battery Tempreture is not being reported correctly.

wavesailor, 2024-07-19:
- Arduino board:  Adafruit ESP32-S2 Feather with BME280 Sensor - STEMMA QT - 4MB Flash + 2 MB PSRAM (LC709203F version)

- Battery: Adafruit Lithium Ion Polymer Battery with Short Cable - 3.7V 420mAh

- Arduino IDE version (found in Arduino -> About Arduino menu):  2.3.2

- Arduino Library: Adafruit_LC709203F version 1.3.4

- Using the LC709203F_demo.ino demo sketch from the examples.
 
- The number reported is the "Battey Temp" field is not correct. I'm assuming it is degrees Celsius. I'm currently indoors and it should be closer to 27 degrees C. I tested the chip temp and it was 31 and battery was 27

Serial output:
```
09:23:44.393 -> Batt_Voltage:3.705	Batt_Percent:13.4	Batt_Temp:8.1
09:23:46.375 -> Batt_Voltage:3.705	Batt_Percent:13.4	Batt_Temp:8.1
09:23:48.406 -> Batt_Voltage:3.705	Batt_Percent:13.4	Batt_Temp:8.1
09:23:50.404 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:23:52.416 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:23:54.390 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:23:56.443 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:23:58.446 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:24:00.419 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:24:02.453 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:24:04.434 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:24:06.478 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:24:08.450 -> Batt_Voltage:3.711	Batt_Percent:13.4	Batt_Temp:8.1
09:24:10.500 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:24:12.491 -> Batt_Voltage:3.708	Batt_Percent:13.4	Batt_Temp:8.1
09:24:14.504 -> Batt_Voltage:3.711	Batt_Percent:13.4	Batt_Temp:8.1
09:24:16.514 -> Batt_Voltage:3.711	Batt_Percent:13.4	Batt_Temp:8.1
09:24:18.493 -> Batt_Voltage:3.711	Batt_Percent:13.4	Batt_Temp:8.1
```

dhalbert, 2024-07-19:
The LC709203F measures battery temperature via an external thermistor that needs to be mounted on the battery. On the  Feather, there is no connection to the thermistor pin on the LC709203F. So the battery temperature can't be measured, and the value you see is just what's seen without a thermistor connected.

I'll add some caveats to the guide about this.

(Info here about the thermistor pin on an external LC709203F breakout: https://learn.adafruit.com/adafruit-lc709203f-lipo-lipoly-battery-monitor/pinouts#other-pins-3071246)

wavesailor, 2024-07-19:
> The LC709203F measures battery temperature via an external thermistor that needs to be mounted on the battery. On the Feather, there is no connection to the thermistor pin on the LC709203F. So the [...]

Okay - Thanks.

closed 2024-07-19 by dhalbert